shufps/ESP-Miner-NerdQAxePlus #42: TMP1075.c: Failure to read temperature does not give more info

jesterhodl, 2024-11-18:
I'm getting this in the logs:
```
₿ (141485) TMP1075.c: Temperature 0: 30.81 C
₿ (141495) TMP1075.c: Failed to read temperature from TMP1075
```

The return value from i2c_master_cmd_begin() in TMP1075_smb_read_word() is not displayed, so I don't know what is the reason :-(

https://github.com/shufps/ESP-Miner-NerdQAxePlus/blob/2f2f45d27b5b3b6c6e8632cdea0b017cb558b59b/main/boards/drivers/TMP1075.cpp#L57

shufps, 2024-11-30:
It is likely to be a hardware problem with the 2nd tmp1075.

I even saw boards from sellers who "forgot" to assemble the tmp1075 on the backside of the board :see_no_evil:

jesterhodl, 2024-12-29:
Yeah, there is no data, but the return value is an error indication. What I was thinking is something to give me a better hint at what's going on. Like this? (i can make make a PR if this is fine)
```
float TMP1075_read_temperature(int device_index)
{
    uint16_t temp_raw;
    int ret = TMP1075_smb_read_word(device_index, TMP1075_TEMP_REG, &temp_raw);

    if (ret == ESP_OK) {
        temp_raw >>= 4;                         // Right-shift to discard the unused bits (12-bit data)
        float temperature = temp_raw * 0.0625f; // Each bit represents 0.0625°C
        ESP_LOGI(TAG, "Temperature %d: %.2f C", device_index, temperature);
        return temperature;
    } else if (ret == ESP_ERR_TIMEOUT) {
        ESP_LOGE(TAG, "Timeout error while reading temperature from TMP1075 (Device %d)", device_index);
    } else if (ret == ESP_ERR_INVALID_ARG) {
        ESP_LOGE(TAG, "Invalid argument passed to I2C command (Device %d)", device_index);
    } else {
        ESP_LOGE(TAG, "Failed to read temperature from TMP1075 (Device %d), error: %d", device_index, ret);
    }

    return 0.0f; // Return a default temperature if reading failed
}

```

shufps, 2025-01-03:
Oh nice, yes sure, I'll merge the PR, looks like a good improvement :ok_hand:

jesterhodl, 2025-01-05:
Ok, cool! That should help debugging some situations.

I'm travelling now, I'll do it mid-next week.

earlynerd, 2025-05-05:
I've discovered all the nerdaxe hardware seems to be lacking i2c bus pullups which may explain this issue. See scope shots on [this nerdoctaxe issue](https://github.com/Patsch91/NerdOCTAXE-Gamma/issues/5)