shufps/ESP-Miner-NerdQAxePlus #44: The flash_size should be 16MB instead of 8MB.

AxisRay, 2024-12-23:
Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**
The `flash_size` should be 16MB instead of 8MB.
[image]
[image]

**To Reproduce**
Manually compile, use merge_bin.sh to merge the firmware files and then flash the firmware.
Then the miner crashed.
```
e (59) flash_parts: partition 4 invalid - offset 0x710000 size 0x400000 exceeds flash chip size 0x800000 e (69) boot: failed to verify partition table e (74) boot: load partition table error!
```

shufps, 2024-12-28:
Oh hmm, thx, I'll investigate :monocle_face:

shufps, 2025-01-03:
Thx for reporting :muscle:

closed 2025-01-03 by shufps