shufps/ESP-Miner-NerdQAxePlus #69: Typo in build images for single chip ultra esp-miner naming.

fromport, 2025-01-29:
The images provided here at github for the BM1366 based ultra nerdaxes are named:
esp-miner-Nerdaxe.bin

I was upgrading from v1.0.21 to v1.0.24 by downloaden the www.bin and the esp-miner-Nerdaxe.bin image
When I wanted to upload the images through the webgui -> settings www.bin. worked great, the firmware option was complaining I hadn't selected the right image.

Took me a while to realize the software is looking for:
esp-miner-Nerd**A**xe.bin  (capital A)

From the the button:
Update Firmware
+ Browse
(esp-miner-NerdAxe.bin)

Should be fairly easy to modify the script that generates the .bin files

shufps, 2025-01-30:
Thx for reporting, this might be connected to some change Bitmaker did by changing a name on the workflow:

https://github.com/shufps/ESP-Miner-NerdQAxePlus/pull/50/files#diff-5c3fa597431eda03ac3339ae6bf7f05e1a50d6fc7333679ec38e21b337cb6721R24

shufps, 2025-01-30:
It needs another change here in order to work properly:

https://github.com/shufps/ESP-Miner-NerdQAxePlus/blob/develop/main/boards/nerdaxe.cpp#L34

This will be used as mining agent too, it will split the `NerdAxe` on public pool into `NerdAxe`s and `Nerdaxe`s

@BitMaker-hub better would be to revert your renaming?

BitMaker-hub, 2025-01-30:
ahh, right I didn't took this into acount.
I think is better just reverting it. 
I'll PR this and some changes I was needing to add due to gamma

BitMaker-hub, 2025-01-30:
fixed here #70

shufps, 2025-02-01:
I reverted the renaming and released new 1.0.24 binaries. So it should work again.

closed 2025-02-01 by shufps