shufps/ESP-Miner-NerdQAxePlus #71: Miners overheating

Zeusln1, 2025-01-31:
Miner have lastest update (v1.0.24).  Standard settings and psu. 

[image]

MyOwn2C, 2025-01-31:
Check your cooling. 
It is working as intended.

Zeusln1, 2025-01-31:
[image]

MyOwn2C, 2025-01-31:
It is working then. 
You just need to disable the overheat message

Zeusln1, 2025-01-31:
I think psu is the problem here.

Zeusln1, 2025-02-01:
I come back with updates: after this incident with overheating, I let the miner overnight to running and now I saw the hash rate is not going up more than 2TH ..  what’s going on? Is the power supply?

MyOwn2C, 2025-02-01:
> I come back with updates: after this incident with overheating, I let the miner overnight to running and now I saw the hash rate is not going up more than 2TH .. what’s going on? Is the power supply?

Without seeing your settings and dashboard, we cannot offer any help. Can’t troubleshoot blind.

Zeusln1, 2025-02-01:
[image]
[image]

Power should be at 54-55w ? Mine is getting stuck at 45-46W (max 1.9-2TH )

Zeusln1, 2025-02-01:
[image]

Zeusln1, 2025-02-01:
> > I come back with updates: after this incident with overheating, I let the miner overnight to running and now I saw the hash rate is not going up more than 2TH .. what’s going on? Is the power supply?
> 
> Without seeing your settings and dashboard, we cannot offer any help. Can’t troubleshoot blind.

[image]
[image]
[image]
[image]

shufps, 2025-02-01:
Is it during the flashing process?

Hmm, I noticed this at least once, it is harmless. But I will investigate how this actually happens.

shufps, 2025-02-01:
Hmm, you should disable the "Invert Polarity" checkbox, to enable it is wrong 99% of the time.

Can you please recheck with disabled invert polarity, disabled auto and set the fan to 100%?

Zeusln1, 2025-02-01:
> Hmm, you should disable the "Invert Polarity" checkbox. Turning it on is wrong 99% of the time. (Maybe I should remove it ...)
> 
> Can you please recheck with disabled invert polarity, disabled auto and set the fan to 100%?

Thank you ! I will check it without that . I will comeback with feedback in max 30 min

Zeusln1, 2025-02-01:
> Thank you ! I will check it without that . I will comeback with feedback in max 30 min

Nothing happens. Still same issue max 2TH and power 44.8W . Fan 100% (2580rpm ) 

[image]
[image]
[image]

shufps, 2025-02-01:
Mmhmm, can you let it run for a while (some hours) and post the `nonce-distribution` from the log?

It's printed into the log when a new share is found.

Zeusln1, 2025-02-01:
Yes I can do.

Zeusln1, 2025-02-01:
> Mmhmm, can you let it run for a while (some hours) and post the `nonce-distribution` from the log?
> 
> It's printed into the log when a new share is found.

₿ (5260094) history: hashrate: 10m:1966.077GH 1h:1822.395GH 1d:109.461GH*
₿ (5260104) history: nonce distribution: 718/711/725/0

Zeusln1, 2025-02-01:
After few restarts the last number of the nonce distribution is still 0. I think is one chip dead or psu broken. In few days I will test with an another PSU.

BitMaker-hub, 2025-02-03:
2TH and power 44.8W. Everything indicates there is less power consumption due to a broken chip

Even the miner can work at 35% fan it's extremelly suggested to run it at 60% minimum

Vinc1347900, 2025-02-09:
Hey guys thanks to this topic, I found out for what I lost 1k of hash. I got the same with the nonce distribution, but I wonder how I can find the dead chip. It's like "nonce distribution: 21878/22720/6400/0" it will be the last chip on the line ? Or do I need to do a bit of research to find out ?

I will order an air gun to try if it's not a faulty solder, but I don't want to break the solder of a healthy chip.

shufps, 2025-03-24:
Ah to the original issue ... this is fixed in 1.0.27.

The automatic fan control reduced PWM to 35% what is too little. This even can lead to blowing the fuse because current consumption increases with temperature what increases current consumption and so on 🥴 

Maybe you want to try with 1.0.27 again?

Or disable the automatic fan control, what I would recommend.

ar63, 2025-04-09:
I noticed that if I keep the fan speed fixed at a high value the RPMs drop (the temperature rises dangerously). If I keep it for example at 25% the RPMs increase (and the temperature obviously decreases). That is, it seems to work the opposite way. Is it a problem with my fan or could it be a bug?
Keeping the speed in automatic creates temperature problems for me, it can't handle it: it configures itself with too low RPMs.

I using the last v.1.0.28.1.

MyOwn2C, 2025-04-09:
Try using the invert fan polarity option, especially when fan RPM works the opposite way. This option flips the fan PWM signal.

ar63, 2025-04-09:
> Try using the invert fan polarity option, especially when fan RPM works the opposite way. This option flips the fan PWM signal.

In my Nerdqaxe I don't see this option. I have this option in my bitaxe, but I don't have the problem there

shufps, 2025-06-03:
In the newer firmware revisions there is an fan polarity auto detection enabled by default so there shouldn't be any polarity issue anymore.

Closing this issue.