shufps/ESP-Miner-NerdQAxePlus #87: Feature request

WantClue, 2025-02-12:
If wifi get's disconnected shut down the buck converter or reset asic jobs and go into an idle state

shufps, 2025-02-13:
The functionality that needs to be implemented feels to complicated for such a rare edge case.

Maybe I won't merge it. Better would be to fix WiFi^^

ar63, 2025-04-22:
I also think this feature is useful. Yesterday I had 3 short blackouts and my 2 miners were offline until I manually restarted them

earlynerd, 2025-06-06:
I've got mine on a smart outlet thing, when the wifi goes down or power goes out briefly, I can count on the esp32 smart outlet device to reassociate with my router when the network is back up. Then I use the smart outlet to power cycle the miner for me, allowing it to also reconnect to my wifi.