shufps/ESP-Miner-NerdQAxePlus #88: Reboot doesn't really check for success

jesterhodl, 2025-02-12:
As I found in #84 reboot function just fires and assumes everything's ok. Although it wasn't OK (401) it was showing a green popup.
I volunteer to take this on.

shufps, 2025-02-13:
Oh nice, can you please hold off a bit? 😅 

I'm currently working on a biiiig UI update - will release it any moment as testing version in the 
https://github.com/shufps/ESP-Miner-NerdQAxePlus/tree/new-ui

branch 😊

shufps, 2025-02-13:
It's released in `v1.0.26-rc1`

jesterhodl, 2025-02-13:
Yeah, sure!
Should the PRs be made to your development branch?

shufps, 2025-02-13:
normally yes, I haven't merged the new UI yet into development because I was worried that there could be some blocket that could require a complete revert 🙈 

It's using a completly new framework now.

So I think it would be a good idea to target the PR for the new-ui branche this time 🙈

shufps, 2025-02-13:
Is fixed here: https://github.com/shufps/ESP-Miner-NerdQAxePlus/commit/be39dce59d30b40873e8e3921b3f4b4fbf89d936

The swarm component implemented it correctly, so it was just copy & paste 😊

closed 2025-02-13 by shufps