shufps/ESP-Miner-NerdQAxePlus #119: Feature request: Firmware for Lilygo T-ETH-Lite

coold-wallet, 2025-03-22:
Hi

Would it be possible to create a firmware version that runs on this Board?
https://lilygo.cc/products/t-eth-lite

I don't need the display for status information as my miner sits my home network rack and i check the webgui regularly.
But i would love to connect my miner via ethernet cable.

Thanks for your time and effort in this project.

shufps, 2025-03-24:
hmm the board is not compatible to the board, would require a big redesign 🤔

MyOwn2C, 2025-03-25:
While anything is possible with open source, selling point of Nerd*Axe is that it runs on WiFi and thus it can be placed anywhere without cable restrictions. 
All the schematics are available so you are free to make your own adaptor PCB or connect using jumper cables. 
You can also modify the source code to fit T-ETH-Lite yourself.

closed 2025-05-08 by shufps