shufps/ESP-Miner-NerdQAxePlus #125: Feature request: Remove device from system info

SirMadcrock, 2025-03-26:
Would it be possible to remove the item “deviceModel”: “NerdQAxe+” from the system info, or freely configurable?

Already discovered several pools that block this device, although it is not a classic nerdminer with a few KH/s.

shufps, 2025-03-26:
The deviceModel on the system info is not what is blocking.

It's the mining agent that is used in the `mining.subscribe` Stratum request.

https://github.com/shufps/ESP-Miner-NerdQAxePlus/blob/develop/components/stratum/stratum_api.cpp#L383-L391

Hmm, I'll add it to my backlog to make it configurable

SirMadcrock, 2025-03-31:
Thanks for the explanation, and especially that you will implement this option. By the way, v1.0.28.1 has been running smoothly on my NerdQAxe+ since yesterday 👌

SirMadcrock, 2025-04-15:
> The deviceModel on the system info is not what is blocking.
>
> It's the mining agent that is used in the `mining.subscribe` Stratum request. [...]

Is there any news on when it will be implemented? Question for a friend 😉