shufps/ESP-Miner-NerdQAxePlus #132: Influxdb support seems to be broken

andreasnanko, 2025-03-28:
```
₿ I (62822) InfluxDB: URL: http://xxxxxxxxxxx:8086/api/v2/buckets?org=miner&name=miner
₿ I (62862) InfluxDB: HTTP GET Status = 200, content_length = 140
₿ W (62862) InfluxDB: Bucket not found
₿ E (62862) influx_task: Bucket not found!
```

So, the check states
- connection ok
- auth ok (200)
- bucket found (200 instead of 404)

But still it say's "Bucket not found"

shufps, 2025-04-03:
Influxdb runs constantly on my NerdQAxes 🤔 

I'm pretty sure it is working.

How does your config look like? Can you post a screenshot please?

shufps, 2025-04-17:
Here is a setup that is working: https://github.com/shufps/ESP-Miner-NerdQAxePlus/tree/develop/monitoring

Closing this issue because influxdb is working.

closed 2025-04-17 by shufps

shufps, 2025-09-14:
maybe this is the issue for you too: https://github.com/shufps/ESP-Miner-NerdQAxePlus/issues/255#issuecomment-3288144090