shufps/ESP-Miner-NerdQAxePlus #138: Feature Request: Network settings

mane, 2025-04-10:
It would be better to have the option to configure network settings manually, rather than being forced to rely solely on DHCP.

MyOwn2C, 2025-04-10:
DHCP is configured via your router. 
Bitaxe has nothing to do with that. 
You can assign static IP at your router.

MyOwn2C, 2025-04-11:
> [@MyOwn2C](https://github.com/MyOwn2C) I know it can be done through the router, however there isn't always the chance to control the router (i.e. not in your own network) and being able to configure the IP address from the device would be handy.

If you have no control of the network or the router, you have no control over what IP you get. 
That is not just Bitaxe, it is for all devices with IP: your phone, laptop, PS5, etc.

mane, 2025-04-11:
@MyOwn2C ok let’s put it this way, imagine you connect the device on a network where you have no access/control to the router but you know that the DHCP has, for instance, a range from .100 to .200, the other IPs are free, you connect the device the network, how do you know which IP address has been assigned to it? Anyway I give up, thanks anyway 🙂
Keep the good work up!

shufps, 2025-04-17:
But if there is a DHCP you get an IP from it 🤔  

Hmm, I guess this is some "won't implement" edge case 🤷‍♂ 

Closing this issue

closed 2025-04-17 by shufps