shufps/ESP-Miner-NerdQAxePlus #145: macAddr is missing from /api/system/info

austrisv, 2025-04-18:
Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**
macAddr missing in /api/system/info output

**To Reproduce**
go to http://<nerdqaxe++IP>/api/system/info

**Expected behavior**
macAddr field present (similarly as for gammas

**Screenshots & Photos**
```
{
    "asicCount": 4,
    "smallCoreCount": 2040,
    "deviceModel": "NerdQAxe++",
    "hostip": "10.10.10.13",
    "power": 81.625,
    "maxPower": 100,
    "minPower": 52,
    "voltage": 11609.38,
    "maxVoltage": 13,
    "minVoltage": 11,
    "current": 7031.25,
    "temp": 57.4375,
    "vrTemp": 55.9375,
    "hashRateTimestamp": 2213227,
    "hashRate": 4800.673,
    "hashRate_10m": 4800.673,
    "hashRate_1h": 4800.673,
    "hashRate_1d": 4800.673,
    "bestDiff": "105M",
    "bestSessionDiff": "8.86M",
    "coreVoltage": 1150,
    "coreVoltageActual": 1144,
    "sharesAccepted": 303,
    "sharesRejected": 1,
    "isUsingFallbackStratum": false,
    "fanspeed": 82,
    "fanrpm": 2292,
    "hostname": "NerdQaxe++",
    "ssid": "myssid",
    "stratumURL": "",
    "stratumPort": 3333,
    "stratumUser": "2",
    "fallbackStratumURL": "public-pool.io",
    "fallbackStratumPort": 21496,
    "fallbackStratumUser": "",
    "frequency": 600,
    "jobInterval": 500,
    "overheat_temp": 70,
    "flipscreen": 0,
    "invertscreen": 0,
    "autoscreenoff": 0,
    "invertfanpolarity": 0,
    "autofanspeed": 1,
    "ASICModel": "BM1370",
    "uptimeSeconds": 2213,
    "lastResetReason": "Software reset",
    "wifiStatus": "Connected!",
    "freeHeap": 5225228,
    "version": "v1.0.28.1",
    "runningPartition": "ota_1"
}
```

**Hardware (please complete the following information):**
 - Bitaxe HW version: NerdQaxe++
 - Bitaxe HW vendor: PM
 - ESP-Miner FW version: 1.0.28.1
 - Hash Frequency: not relevant
 - Voltage: not relevant
 - Pool URL, Port, User: not relevant

**Additional context**
would make a testing/qc life much easier being able to uniquely identify the devices 🫶

shufps, 2025-04-24:
It's not missing, it just hasn't been added yet 🤪 

Will add it asap 😊

shufps, 2025-04-25:
added in #156