shufps/ESP-Miner-NerdQAxePlus #150: UI/UX: user info field of the dashboard

zcht, 2025-04-18:
**Describe the bug**

At the break point of 992 px, the BTC address and suffix are not displayed completely. A scroll bar appears here, which is not ideal from a UI/UX perspective.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Dashboard'
2. Reduce the viewport to 992px

Possible solutions:
1. Display the first and last 5 characters + suffix
2. Reduce font size significantly
3. Adjust break point. 

**Screenshots & Photos**
[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: NerdQaxe++
 - ESP-Miner FW version: 1.29
 - Pool URL, Port, User: public pool on umbrel

koin-x, 2025-04-20:
This does not happen for me on my monitor. Might just happen because your monitor is smaller.

zcht, 2025-04-20:
I have three monitors and the screenshot was taken on a 32” screen ;)

shufps, 2025-04-21:
interesting, thx for the bug report, I'll investigate 🧐

closed 2025-05-08 by shufps