shufps/ESP-Miner-NerdQAxePlus #166: Miner sends duplicate shares

adriankirchner, 2025-05-11:
**Describe the bug**
I'm using a private instance of [benjamin-wilson/public-pool](https://github.com/benjamin-wilson/public-pool/). After updating the pool to include [this commit](https://github.com/benjamin-wilson/public-pool/commit/e1b88207d3c5f6f9496dd63f39582d12ad152fab), it started rejecting over 1000 shares per miner per day (see Grafana screenshot).

The referenced commit introduces detection of duplicate share submissions. In the miner logs, you can see that the miner sometimes submits the same share twice:
```
₿ I (112088641) asic_result: Job ID: 50 AsicNr: 0 Ver: 2FB0E000 Nonce 32DC0284 diff 4907.0 of 1024.
₿ I (112088641) history: hashrate: 10m:3000.558GH  1h:3105.259GH  1d:3055.777GH 
₿ I (112088651) history: nonce distribution: 18595/20587/17721/19361
₿ I (112089321) asic_result: Job ID: 68 AsicNr: 1 Ver: 26A6C000 Nonce BD3F0448 diff 16154.2 of 1024.
₿ I (112089321) stratum_api: tx: {"id": 8553, "method": "mining.submit", "params": ["REDACTED.nerdqaxe", "aa6e", "0001742f", "682135a9", "bd3f0448", "06a6c000"]} <-- duplicate share
₿ I (112089341) history: hashrate: 10m:3004.401GH  1h:3105.881GH  1d:3055.803GH 
₿ I (112089341) history: nonce distribution: 18595/20588/17721/19361
₿ I (112089351) asic_result: Job ID: 68 AsicNr: 1 Ver: 26A6C000 Nonce BD3F0448 diff 16154.2 of 1024.
₿ I (112089361) stratum_api: tx: {"id": 8554, "method": "mining.submit", "params": ["REDACTED.nerdqaxe", "aa6e", "0001742f", "682135a9", "bd3f0448", "06a6c000"]} <-- duplicate share
₿ I (112089371) stratum task: rx: {"id":8553,"error":null,"result":true}
₿ I (112089381) history: hashrate: 10m:3011.496GH  1h:3107.064GH  1d:3055.853GH 
₿ I (112089391) stratum task: message result accepted
₿ I (112089391) history: nonce distribution: 18595/20589/17721/19361
₿ I (112089431) stratum task: rx: {"id":8554,"result":null,"error":[22,"Duplicate share",""]} <-- reject message
₿ W (112089431) stratum task: message result rejected
₿ I (112089471) TPS53647.c: raw temp: eaba
```

This behavior occurs consistently across all four NerdQAxe+ miners in the inventory.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect miner to a public-pool instance including this commit eg. https://web.public-pool.io/
2. Scan log for messages containing `Duplicate Share`

**Expected behavior**
The miner should not submit more than 1000 duplicate shares per day. This should be a rare occurrence, e.g., in cases of connection failure.

**Screenshots & Photos**
[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: NerdQAxe+ (BM1368)
 - Bitaxe HW vendor: miningwholesale.eu
 - ESP-Miner FW version: v1.0.29.1 and v1.0.28.1
 - Hash Frequency: 640
 - Voltage: 1350
 - Pool URL, Port, User: my pool is private but I guess this is also happening with the official instance: https://web.public-pool.io/

**Additional context**
This does not happen with Bitaxe Gamma 601 (Firmware [bitaxeorg/ESP-Miner](https://github.com/bitaxeorg/ESP-Miner))

adriankirchner, 2025-05-27:
Yes, the miner is overclocked. Reducing the interval by 10ms and 100ms did not result in any noticeable change, but I will continue testing.

adriankirchner, 2025-05-27:
Maybe helpful: our NerdQAxe++ that arrived today, with the same firmware version, does not have this problem. Same private pool and overclocked (5.4 TH/s @ 1180 mV / 650 MHz). It’s possible that this only affects the NerdQAxe+.

shufps, 2025-05-28:
Hmm have you rebooted after changing the job interval time?

adriankirchner, 2025-05-28:
Even though it wasn’t necessary, I restarted the miner after every change to the job interval.

0x-fran, 2025-05-28:
The BM1368 are real drama queens ... 
Mine sulked above 1.7 TH/s on stock settings - only calmed down after some therapy at 1350 mV and 550 MHz.
No duplicate drama here though… but I also don’t overclock like it’s a benchmark contest 😅

adriankirchner, 2025-05-28:
Indeed, the NerdQAxe++ is running smoothly with the BM1370. Feel free to close the issue.

closed 2025-05-31 by shufps