shufps/ESP-Miner-NerdQAxePlus #183: Stratum TCP socket leak after extended uptime (BM1370 v1.0.29.1)

jmbinette, 2025-06-05:
Description:
When running my NerdQAXE++ BM1370 on v1.0.29.1, stratum connections work perfectly after reboot.
After ~24 hours of uptime, pool connections start silently failing: socket connects successfully, but no stratum response is received from the pool (Parasite.wtf in this case).
Rebooting my router temporarily fixes the problem for another 24h cycle.
This strongly suggests an issue with socket handling or NAT exhaustion caused by stale or unclosed sockets accumulating over time.

Observed Behavior:

Miner starts successfully, connects to stratum+tcp://parasite.wtf:42069, performs subscribe/authorize sequence normally.

After ~18-30 hours of uptime, connections silently hang after authorize is sent.

Socket creation and connection (stratum task: Connected) still succeed, but pool never responds.

WiFi and network are otherwise fully functional.

Rebooting the router fully restores normal operation temporarily.

Issue repeats consistently every 24h or so.

Suspected Root Cause:

TCP sockets may not be properly closed during reconnect attempts or error handling.

Old sockets accumulate in the router's NAT table (or possibly inside lwIP on the ESP32 itself).

After enough failed reconnects, NAT tables are exhausted, new sessions behave erratically, and stratum handshake breaks.

Firmware Version:
v1.0.29.1 (BM1370)

Hardware:
NerdQAXE++ BM1370 (ESP32-S3)

Pool:
Parasite.wtf (stratum+tcp://parasite.wtf:42069)

shufps, 2025-06-06:
Hmm interesting observation, thx a lot.

Network has always been an issue and it seems it never has been fully solved.

Hmm, I'll battle-test the pool failover, maybe I see something weird happening 🤔

shufps, 2025-06-06:
Hmm so far it looks very fine:
```
$ while true; do echo $( date ); timeout 120 ncat -l 6666 --sh-exec "ncat solo.ckpool.org 3333" --keep-open; sleep 120; done
Fr 6. Jun 08:17:25 CEST 2025
Fr 6. Jun 08:21:25 CEST 2025
Fr 6. Jun 08:25:25 CEST 2025
Fr 6. Jun 08:29:25 CEST 2025
Fr 6. Jun 08:33:25 CEST 2025
Fr 6. Jun 08:37:25 CEST 2025
Fr 6. Jun 08:41:25 CEST 2025
Fr 6. Jun 08:45:25 CEST 2025
Fr 6. Jun 08:49:25 CEST 2025
Fr 6. Jun 08:53:25 CEST 2025
Fr 6. Jun 08:57:25 CEST 2025
Fr 6. Jun 09:01:25 CEST 2025
Fr 6. Jun 09:05:25 CEST 2025
Fr 6. Jun 09:09:25 CEST 2025
Fr 6. Jun 09:13:25 CEST 2025
Fr 6. Jun 09:17:25 CEST 2025
Fr 6. Jun 09:21:25 CEST 2025
Fr 6. Jun 09:25:25 CEST 2025
Fr 6. Jun 09:29:25 CEST 2025
Fr 6. Jun 09:33:25 CEST 2025
Fr 6. Jun 09:37:25 CEST 2025
Fr 6. Jun 09:41:25 CEST 2025
Fr 6. Jun 09:45:25 CEST 2025
Fr 6. Jun 09:49:25 CEST 2025
Fr 6. Jun 09:53:25 CEST 2025
Fr 6. Jun 09:57:25 CEST 2025
Fr 6. Jun 10:01:25 CEST 2025
Fr 6. Jun 10:05:25 CEST 2025
Fr 6. Jun 10:09:25 CEST 2025
```
It runs `ncat` as proxy to `solo.ckpool.org` for 2 minutes, kills the proxy and forces failover for 2min and then it starts the proxy again.

May I ask what Stratum pool you are using?

jmbinette, 2025-06-07:
**Follow-up after deeper analysis with my network logs:**

The root issue seems linked to how NerdAxe++ handles reconnect cycles when pool connectivity is unstable or interrupted. On my Asus AX88U router (Broadcom-based NAT), I'm seeing repeated log entries like:

nf_conntrack: expectation table full

**Observations:**

- NerdAxe++ aggressively reconnects very quickly when pool connections drop.
- The NAT state is not always fully torn down before reconnecting.
- The router keeps filling up the *expectation table* (Broadcom hardware NAT implementation via `BCM-NAT`), even if overall conntrack count remains healthy.
- Eventually, this blocks new outgoing connections even though all other devices on my network stay fully functional.

**Important context:**

- This is **not strictly a router bug** (though Broadcom NAT implementations have known limitations here).
- I have 63 devices active on the network, only the miner exhibits this issue.
- The miner’s fast connection retries appear unique in exhausting this NAT table.

---

**Possible firmware-side improvements:**

- Introduce exponential backoff or minimum reconnect interval after disconnects.
- Ensure full TCP socket teardown to avoid TIME_WAIT / lingering sessions.
- Actively close or clean up NAT sessions before reconnecting.
- Use TCP keepalives earlier to maintain NAT bindings.
- Avoid UDP NAT expectations if applicable.
- Make reconnect delays user-configurable for more flexibility.

---

shufps, 2025-06-07:
I'm pretty sure we don't leave anything opened or uncleaned but I'll review the code (once again^^).

Hmm, I wonder if esp-idf update could help because the Nerd*Axe firmware is only using high level functions provided by the SDK, maybe the underlaying lwip has bugs idk and updating the build process with using a newer esp-idf version could fix it 🤔 

Thx a lot for this analysis - networking always has been a pain and I think we are a step closer to the problem 🙈

shufps, 2025-06-07:
I asked gemini 2.5 pro for problems regarging cleanup, error handling, socket closing and such on `straum_task.cpp` and `stratum_api.cpp` and it responded with:

```
Final Conclusion
The code is exceptionally well-written with respect to socket lifetime management. The developers have clearly considered the various failure modes of a long-lived network connection and have implemented a robust, centralized cleanup mechanism that is triggered by any of these failures. You can be confident that in the scenarios presented, socket descriptors are not being leaked.
```

Hmm ...

0x-fran, 2025-06-07:
We could implement a TCP Keepalive within function `setupSocketTimeouts`:

```
    // Enable TCP Keepalive
    int enable = 1;
    if (setsockopt(sock, SOL_SOCKET, SO_KEEPALIVE, &enable, sizeof(enable)) < 0) {
        ESP_LOGE(m_tag, "Failed to enable SO_KEEPALIVE");
        return false;
    }

    // Configure Keepalive parameters
    int keepidle = 10;   // Start sending keepalive probes after 10 seconds of inactivity
    int keepintvl = 5;   // Interval of 5 seconds between individual keepalive probes
    int keepcnt = 3;     // Disconnect after 3 unanswered probes

    if (setsockopt(sock, IPPROTO_TCP, TCP_KEEPIDLE, &keepidle, sizeof(keepidle)) < 0) {
        ESP_LOGE(m_tag, "Failed to set TCP_KEEPIDLE");
        // This might not be critical, so we could just log a warning and continue
    }
    if (setsockopt(sock, IPPROTO_TCP, TCP_KEEPINTVL, &keepintvl, sizeof(keepintvl)) < 0) {
        ESP_LOGE(m_tag, "Failed to set TCP_KEEPINTVL");
    }
    if (setsockopt(sock, IPPROTO_TCP, TCP_KEEPCNT, &keepcnt, sizeof(keepcnt)) < 0) {
        ESP_LOGE(m_tag, "Failed to set TCP_KEEPCNT");
    }

    ESP_LOGI(m_tag, "TCP Keepalive enabled: idle=%ds, interval=%ds, count=%d", keepidle, keepintvl, keepcnt);

```

But there are two other things:
1. The `connect() `system call is passed `sizeof(struct sockaddr_in6)` for an `AF_INET` (IPv4) socket. However, the address structure being used is `struct sockaddr_in`, which causes the kernel to read past the bounds of the `dest_addr` struct on the stack.

Fix: 
```
-	int err = ::connect(sock, (struct sockaddr *) &dest_addr, sizeof(struct sockaddr_in6));
+	int err = ::connect(sock, (struct sockaddr *) &dest_addr, sizeof(dest_addr));
```

2. Not network related, but the `memset()` call intended to clear the `m_stratum_api_v1_message` struct before parsing a new message uses `sizeof(m_stratum_api_v1_message)`. We should change the size argument to `sizeof(StratumApiV1Message) `to ensure the entire structure is properly zeroed out before each use

Fix:
```
-	memset(m_stratum_api_v1_message, 0, sizeof(m_stratum_api_v1_message));
+	memset(m_stratum_api_v1_message, 0, sizeof(StratumApiV1Message));
```

shufps, 2025-06-07:
Hmm, thx a lot - oh yes, I'm fixing it. Maybe it repairs something right away.

jmbinette, 2025-06-08:
### Update after testing v1.0.30-rc1-idf-5.5

- Flashed and running v1.0.30-rc1-idf-5.5. NAT passthrough disabled (RTSP/SIP/H323 off).
- After miner reboot: miner booted, LCD showed `.110`, DHCP assigned `.110`, and miner connected properly to pool.
- Rebooted router (without rebooting miner):
  - DHCP re-assigned `.109`.
  - LCD still showed `.110`, but miner successfully reconnected automatically to pool.
  - NAT mapping confirmed active via conntrack.
- Performed manual test by disabling and re-enabling 2.4GHz radio (`wl -i eth6 down` / `wl -i eth6 up`):
  - After interface came back online, miner failed to reconnect to pool.
  - Error `selected pool not connected` was shown.
  - Pool connection did not recover automatically after 2.4 GHz recovery.
  - Manual reboot of miner was needed to restore pool connection.
- Summary:
  - DHCP/NAT behavior looks improved after router reboot.
  - LCD IP desync issue still present.
  - Pool connection recovery after WiFi loss (2.4GHz toggle) remains an issue.

shufps, 2025-06-08:
Mmmhmmm, the LCD IP is only updated once at boot. 

Why is your DHCP server assigining new IPs each time you reboot? 🤔 

But I guess this is something of the easier fixable things^^

jmbinette, 2025-06-09:
Thanks for the quick reply!

Yes — DHCP behavior is normal here: my router sometimes reassigns IPs after reboots depending on lease state (but miner stays functional as long as connection is valid). The main problem remains on pool reconnection after WiFi drop:

Observations during disconnect
Miner remains fully online on LAN (verified via ARP on router: 192.168.1.109 with correct MAC b4:3a:45:a2:35:5c).

LAN side: DHCP lease stable, IP address unchanged.

Router (AsusWRT Merlin RT-AX88U 3004.388.9_2) shows no NAT table entry for port 42069 at time of disconnect:

conntrack -L -p tcp | grep 42069
:No active entries

Device logs:

Keeps producing ASIC results.

Periodically prints stratum-manager: selected pool not connected.

Socket check shows Socket is still connected. but pool is not receiving work.

After failure, miner eventually tries fallback pool as designed.

LCD keeps showing previously assigned IP (sometimes inconsistent with router ARP after DHCP renewals).

No watchdog or self-recovery occurs automatically after pool disconnect.

Additional note (from wl eth6 test)

After WiFi interface down/up (wl -i eth6 down && wl -i eth6 up), miner reconnects to LAN but never re-establishes stratum pool connection.

It seems socket state survives interface drop but pool session is lost. No reconnect/reopen sequence is triggered.

Problem could be related to how socket errors / TCP state are monitored after underlying transport (WiFi) interruptions.

Happy to run additional debug builds if needed!

0x-fran, 2025-07-13:
... merged and released in 1.0.31-rc1. Should be available on the webflasher - or just check here: https://github.com/shufps/ESP-Miner-NerdQAxePlus/releases/tag/v1.0.31-rc1