shufps/ESP-Miner-NerdQAxePlus #194: Stratum not connected. Skipping ping... |

JohnnyPessa, 2025-06-12:
Hello,
Today I received the NerdQAxePlus miner, after a quick connection to my home network I set solo.ckpool.org, Port: 3333 and my btc wallet.0 as stratum user, updated the miner to version 1.0.30 via the dashboard and after 3 hours the performance is still at 0 GH/s. Realtime logs still show me this message: W (1109144) ping task: Stratum not connected. Skipping ping... Pool Information: Stratum not connected. Pool info unavailable. Can you please help me how to set this miner up correctly? Or how to reset it to factory settings. Otherwise it seems to work (79W, 54 Celsius) and communicate normally, it just doesn't mine... I've tried other pool settings (eusolo, default poll.powermining.io, etc), but nothing helps.  Thank you for your help.

[image]
[image]

0x-fran, 2025-06-13:
This message is displayed only if the `stratum task` is unable to establish a connection to the specified pool (either primary or fallback). It seems that, even though you are connected to Wi-Fi, your network is either not connected to the internet or something else might be causing the issue (e.g., router problems). On the System Page, you can filter the logs by entering keywords such as 'stratum task' or 'DNS' to investigate what might be happening.

Have you tried to enter the IP instead of the hostname? For `solo.ckpool.org` it's `15.204.102.129`

JohnnyPessa, 2025-06-13:
> This message is displayed only if the `stratum task` is unable to establish a connection to the specified pool (either primary or fallback). It seems that, even though you are connected to Wi-Fi, [...]

Thank you for your help. You were right. My Asus router blocked Bitcoin mining through built-in AI software protection, so I turned it off (some components) and hooray, everything is working good now...