shufps/ESP-Miner-NerdQAxePlus #199: Wifi RSSI (dbm values)

shufps, 2025-06-18:
request:
```
would it be possible that in one of the next firmware updates, the Wifi RSSI (dbm values) will also be made available in the API for the NerdQAxe+, similar to what is now the case with the Bitaxe?
```

0x-fran, 2025-06-19:
Here we go — happy to hear your thoughts! 🤔

---

Bitaxe exposes `wifiRSSI` in the system info response via `ISystemInfo.ts` as:

```
wifiRSSI: number,
```

---
**Current state (ready for PR):**

**REST API:**
```
curl http://YOUR-BITAXE-IP/api/system/info | jq     
[…]
  "wifiRSSI": -45,
  "wifiStatus": "Connected!",
[…]
```

**System Log:**
```
₿ I (44039507) WIFI_RSSI: Current RSSI: -45 dBm
```

**WebUI preview:**

[image]

Includes compact inline display with 4 bars and a quality tooltip:

```
public getRssiTooltip(rssi: number): string {
  if (rssi <= -85) return 'Signal strength: Very weak (≤ -85 dBm)';
  if (rssi <= -75) return 'Signal strength: Weak (≤ -75 dBm)';
  if (rssi <= -65) return 'Signal strength: Moderate (≤ -65 dBm)';
  if (rssi <= -55) return 'Signal strength: Strong (≤ -55 dBm)';
  return 'Signal strength: Excellent (> -55 dBm)';
}
```

Let me know if you’d like a patch — I’m happy to open a PR! 🙌

shufps, 2025-07-04:
working nicely, closing this issue 🙌

closed 2025-07-04 by shufps