shufps/ESP-Miner-NerdQAxePlus #226: Feature Request - Automatic WIFI connection retry after power failure.

pTennn, 2025-07-29:
Hi Team.

After having a big storm, our power has been surging a lot. What i have noticed is my NerdQaxes++ will hang with a WIFI connection error on screen. 

The bitaxes will connect up fine. 

What i think is happening is the NerdQaxes are booting up faster than what my router does and stopped trying to connect as the hostname isn't available at that time.

Simply me disconnecting the nerdqaxes from the power and back on again fixes this issue - but more of a pain when it happens when i am not around.

hermeticvm, 2025-08-07:
Most likely a dupe of: https://github.com/shufps/ESP-Miner-NerdQAxePlus/issues/87

mfinnz, 2025-08-14:
I'd also like to vote for this as a feature request. To overcome the poor WiFi signal of the Lilygo board I am using a small WiFi repeater which can be powered with the 5V pin. Unfortunately as the repeater, a GL-AR150, takes 30-45 secs to power up the Nerdqaxe gives up on connecting to the Wifi and just sits there. Waiting an hour for it to time out and connect is a bit long really.

shufps, 2025-09-10:
should be fixed now with `v1.0.33-test1`.

It's available in the webflashed (with pre-releases enabled).