shufps/ESP-Miner-NerdQAxePlus #233: Bug(?): Fan Speed Always at 100% Despite PID Setting

ceesvanegmond, 2025-08-14:
Hi,

I'm encountering an issue with one of my NerdQAxe++ units where the fan speed consistently runs at 100%, even though I've configured the fan controller to use PID mode with a target temperature of 60 °C.

Based on how PID control should work, I expect the fan speed to adjust dynamically to maintain the ASIC temperature near the target. However, in this case, the ASIC temperature remains below 60 °C, yet the fan speed is still being set to 100%. This behavior seems inconsistent with the intended function of PID control.

The unit is running firmware version v1.0.31, and the settings are identical to two other NerdQAxe++ units I have in the same room, using the same stock cooling setup. Those units behave as expected, with fan speeds adjusting appropriately based on temperature.

I've attached screenshots showing the temperature readings, fan speed, system logs, and configuration settings to help illustrate the issue.

Model: NerdQAxe++ (BM1370)
Firmware: v1.0.31

Is this a bug in the firmware, or could I be misunderstanding how the PID feature is supposed to work?

Thanks in advance for your support!

[image]
[image]
[image]

kukulle-hood, 2025-08-17:
Hi Cees, I had this also once, but then I have reinstalled the firmware AND the www.bin and it worked again. I am now running the beta v1.0.32 test1 version and it also worx fine.

shufps, 2025-08-22:
ASIC and VR-Temp is used for the PID. It will try to keep the higher of both values at 60°C.

Your VR-Temp is at 65°C so it will run at 100% because there is no extra fan that is cooling the VR.

But there is the option to plug another (eg 40mm or a bigger at the backside - I have seen different solutions) fan on the 2nd connector and cool the VR.

Closing this issue because it's not a bug

closed 2025-08-22 by shufps