shufps/ESP-Miner-NerdQAxePlus #235: Feature Request - Settings inside "Danger zone"

kukulle-hood, 2025-08-17:
SETTINGS - DANGER ZONE

Fan-Controller
•	Is it possible to split the 2 existing fan-controller in fan controller 1 and 2 (or upper socket and lower socket or M1 and M2)?
•	Can you implement a kind of Fan Calibration feature for each fan?
-The idea behind, not only me alone but seen in a lot of threads and images that people where sharing that they are using different kind of cooler-fans.
Besides the fact, that mostly the speed of the front side cooler is much higher needed then the one on the backside (or somewhere else), mostly the type of coolers and basic speed ranges are also different. I found out e.g. that my cooler at the front side reaches its max speed at about 80% of manual setting. So going to 100% does not change something. I think a calibration for each fan would be great and brings in realistic speed feedbacks.

Thanx,
kukulle

kukulle-hood, 2025-09-06:
Does this open source software have scripts which can be used to be implemented here in our system?
[https://github.com/Rem0o/FanControl.Releases](url)

shufps, 2025-09-07:
Yes they can be controlled separately, and there is a open PR for the 2nd fan.