shufps/ESP-Miner-NerdQAxePlus #255: InfluxDB Connection Issue

stefanoneri, 2025-09-05:
Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**
I cannot connect my NerdAxe (running Software Version 1.0.31) to a self-hosted InfluxDB2 instance. Here is what the miner is logging:
```
₿ I (18307) InfluxDB: Successfully connected to InfluxDB at http://192.168.0.195:8086
₿ I (18317) InfluxDB: URL: http://192.168.0.195:8086/api/v2/buckets?org=home&name=nerdqaxeplus
₿ I (18337) InfluxDB: HTTP GET Status = 401, content_length = 55
₿ E (18337) InfluxDB: HTTP GET failed with status code: 401
₿ E (18347) influx_task: Bucket not found!
₿ I (18347) InfluxDB: Creating bucket at URL: http://192.168.0.195:8086/api/v2/buckets
₿ I (18357) InfluxDB: Looking up orgID via: http://192.168.0.195:8086/api/v2/orgs?org=home
₿ I (18377) InfluxDB: Org lookup response: {"code":"unauthorized","message":"unauthorized access"}
₿ E (18387) InfluxDB: Failed to extract org ID from JSON
₿ E (18387) InfluxDB: Could not resolve org ID
₿ E (18387) influx_task: Bucket couldn't be created!
```
What I did to troubleshoot:

- recreated the API tokens twice without success
- manually created the bucket without success
- cross-checked org - which is valid

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'InfluxDB'
2. Enter Influx URL, Port, Token, Bucket, Org, Prefix
3. Click on 'SAVE'
4. Click on 'RESTART'
5. See error

**Expected behavior**
Miner should be able to connect to InfluxDB and stream data there in the bucker 'nerdqaxeplus'

**Screenshots & Photos**

[image]

[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: NerdAxe (BM1366)
 - Bitaxe HW vendor: AliExpress
 - ESP-Miner FW version: v1.0.31
 - Hash Frequency: 525MHz
 - Voltage: 1.26
 - Pool URL, Port, User: pool.solomining.de 3333

shufps, 2025-09-05:
Mmhmmm ...

does this work with your token?

```
curl -G http://192.168.0.195:8086/api/v2/buckets?org=home \
  --header "Authorization: Token YOUR_TOKEN_HERE" 

```

stefanoneri, 2025-09-05:
Nope, it is not working - what could be the error?

```
root@thor:~# curl -i -X GET http://192.168.0.195:8086/api/v2/buckets?org=home&name=nerdqaxeplus -H "Authorization: Token xxx-xxx=="
[1] 2389557
-bash: -H: command not found
root@thor:~# HTTP/1.1 401 Unauthorized
Content-Type: application/json; charset=utf-8
X-Influxdb-Build: OSS
X-Influxdb-Version: v2.7.12
X-Platform-Error-Code: unauthorized
Date: Fri, 05 Sep 2025 12:21:24 GMT
Content-Length: 55

{"code":"unauthorized","message":"unauthorized access"}
```

stefanoneri, 2025-09-05:
I cross checked influxdb logs: 
`ts=2025-09-05T12:18:19.842575Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="token required"
ts=2025-09-05T12:19:37.584122Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="token required"
ts=2025-09-05T12:20:12.812295Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="token required"
ts=2025-09-05T12:20:56.114802Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="token required"
ts=2025-09-05T12:21:24.761910Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="token required"
ts=2025-09-05T12:30:32.155434Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="authorization not found"
ts=2025-09-05T12:30:32.206122Z lvl=info msg=Unauthorized log_id=0ynRwvA0000 error="authorization not found"`

shufps, 2025-09-05:
ah what version is your influxdb? but I guess it's 2.x because it has a org, right?

shufps, 2025-09-07:
Ok 👍 

Hmm can you try this setup here pls?

https://github.com/shufps/ESP-Miner-NerdQAxePlus/tree/develop/monitoring

stefanoneri, 2025-09-13:
Success!
I also disabled InfluxDB - made a restart, added the token again and made another restart, then enabled influxdb: 

₿ I (18343) InfluxDB: URL: http://192.168.0.195:8086/ping
₿ I (18393) InfluxDB: Successfully connected to InfluxDB at http://192.168.0.195:8086
₿ I (18393) InfluxDB: URL: http://192.168.0.195:8086/api/v2/buckets?org=home&name=nerdqaxeplus
₿ I (18423) InfluxDB: HTTP GET Status = 200, content_length = 872
₿ I (18443) InfluxDB: URL: http://192.168.0.195:8086/api/v2/query?org=home
₿ I (18453) InfluxDB: Query JSON: {"query":"from(bucket:\"nerdqaxeplus\") |> range(start:-1y) |> filter(fn:(r) => r._measurement == \"mainnet_stats\") |> last()"}

shufps, 2025-09-14:
k, thx - I guess I'll have to check what the problem is for requiring such workaround 🤔

shufps, 2025-09-21:
Ah yes but on the settings page of Influx it states that you need to reboot the device after changing 😁