shufps/ESP-Miner-NerdQAxePlus #257: Bug - BTC Price & Global Mining Stats

koin-x, 2025-09-07:
BTC Price and Global Mining stats not showing up on NerdAxe and NerdQAxe - Started happening yesterday. Did the most recent update and still no fix. 

I reset all miners because they were showing different prices and after restart they all just say 0 now for the price and for the GMS on 3 different miners

andypeppa, 2025-09-07:
Confirmed behavior on x2 NQA++ running v1.0.32

shufps, 2025-09-07:
Hmm, it's working on my device 🙈 

Is it still not working on yours?

andypeppa, 2025-09-07:
> Hmm, it's working on my device 🙈
> 
> Is it still not working on yours?

Both of my NQAxes still throwing the same error block in OP

kukulle-hood, 2025-09-08:
@koin-x @andypeppa I had the same problem with missing indication in the screen as well the wrong firmware number was indicated. I simply restarted the update process by making sure 1st starting with the firmware file "esp-miner-NerdQAxe++.bin", waiting for restart and then following with the www.bin
For me it worked fine.
Might be after Firmware update that you get a special one pager website indication from the NerdQAxe OS, that you have to install the www.bin seperately. Simply follow the instructions.

koin-x, 2025-09-10:
still no fix.... 😑

shufps, 2025-09-10:
Btw sometimes on boot I see errors in the log and sometimes I don't. Maybe a rate limiting issue.

Do you have more than one device?