shufps/ESP-Miner-NerdQAxePlus #258: esp-miner-NerdQAxe++.bin

fatfoxshade, 2025-09-07:
When you try to flash the latest esp-miner-NerdQAxe++.and bin

v1.0.32 esp-miner-NerdQAxe++.bin 

The version of firmware that displays after the load still comes up as the prior version after the flash

v1.0.31

But the settings will no longer display or show - They just display as a window message "still loading"

I had to revert back to the v1.0.31 esp and bin to restore functionality

kukulle-hood, 2025-09-08:
Hello @fatfoxshade , I once had the same problem, but the reason was very clear. It was my that I first installed the www.bin, what was wrong. After Installing the Firmware first again ( esp-miner-NerdQAxe++.bin ) at my NerdQAxe, all worked fine again.
I also do not have problems with missing information in the T-Display 
Cheers, kukulle-hood

fatfoxshade, 2025-09-10:
When I apply the flash. it says it’s still miner-NerdQAxe++ v1.0.31 afterwards.