shufps/ESP-Miner-NerdQAxePlus #263: Feature Request - Extranounce Subscribe

kukulle-hood, 2025-09-10:
Hello Developers,

I am not too deep in the functionalities of getting better Hashrates with the NerdQAxe++, but in the Bitaxe there is the possibility given, to activate the "ExtraNounce1" via a given checkbox in the settings.

[image]

Is this something what will be implemented soon?

Thanx for info.

kukulle

kukulle-hood, 2025-09-14:
Hello Thomas,

if you join https://bowserlab.ddns.net:8080/ on solo-mode and switch over then to the Wallet tab, you can see your machine running. When entering your wallet address. Half way down yout see this indication (red) when extranounce is activated

[image]

shufps, 2025-09-14:
perfect, thx! I'll try that

shufps, 2025-09-14:
merged, will be in `v1.0.33`

closed 2025-09-14 by shufps