shufps/ESP-Miner-NerdQAxePlus #271: PSU error – Guru Meditation #00000015 on NerdQaxe++ miner

janix12, 2025-09-14:
Hello!

I’m having a hardware issue with my NerdQaxe++ miner.
It shows the following error:

PSU error - Guru Meditation #00000015

I’ve attached the log.

[NQ_log.txt](https://github.com/user-attachments/files/22321289/NQ_log.txt)

Could you please help me figure out what component needs to be replaced?

Thanks in advance,
János

shufps, 2025-09-17:
> I (8060) TPS53647.c: Initializing the core voltage regulator
> I (8060) TPS53647.c: Device Code: a888
> E (8060) TPS53647.c: ERROR- cannot find TPS53647 buck controller
> ```
> 
> It looks suspicious that the input power voltage is shown as 5V on the display. Could it be a hardware issue?

hmm it reports weird things:

> power_management: vin: 1600.00, iin: 0.00, pin: 0.00, vout: 0.00, iout: 0.00, pout: 0.00, vr-temp: 31.62

have you plugged 1600V on it? That might explain the issue you are seeing 🤪  Sorry, just kidding.

Is it a new device? Has it worked before?

kukulle-hood, 2025-10-05:
@janix12 Good Morning to you.
Any News from your side? Did you manage to t´get your device running again?
Robert

shufps, 2025-11-15:
> Unfortunately, one of the ASIC chips is shorted.

Probably not the ASICs. They look shorted because they have a super low resistance, so when measuring it, they look shorted but they are not.  🙈 

The only hardware fault I saw so far was a short in one of the CSD chips.

Important: If it's one of the CSDs you will measure something like a short on the +12V (and a blown fuse). If an ASIC is shorted you won't see a short on +12V

Unfortunately it's impossible to say which one without removing the inductors 🙈 

After removing the inductors each CSD chip can be measured if it's shorted.

If one is shorted it needs replacement 🙈 

If you have still warranty from your seller, better return it 🙏