shufps/ESP-Miner-NerdQAxePlus #277: Feature Request - Average Efficiency

alucine, 2025-09-19:
I'd like to suggest incorporating a feature that's already available in the Bitaxe Gamma panel into the NerdQaxe++ panel. Currently, it displays the live efficiency value alongside the expected efficiency and average efficiency. It would be fantastic to extend this with 1-hour and 1-day efficiency metrics as well. Thank you, and great work

andypeppa, 2025-09-20:
Agreed. I am using influxdb/grafana to track these numbers over time to understand how it my hash rate and efficieny have trended over time, but showing these numbers at a glance would be a welcome addition on the main dashboard.

shufps, 2025-09-21:
v1.0.33 leverages the BM1370’s hardware hash counters for accurate hashrate reporting.
Instead of estimating from found shares, hashrate is now computed directly.
As a result, the efficiency metric no longer needs 1 h or 1 d averages.

Closing this issue. 😊

closed 2025-09-21 by shufps

alucine, 2025-09-21:
Thanks for your reply, is great that the hashrate will be exact and not computed. But, one home miner the most important is efficiency, and average efficiency is a must. Sorry if don't understand your previous message

shufps, 2025-09-21:
I wanted to say that efficiency is power / hashrate and when we know hashrate "exact" we also know efficiency.

Averages are often just used to even out uneven estimations but we can actually measure efficiency directly know 🤔 

Do you agree or did I miss something?

alucine, 2025-09-21:
The efficiency is power / hashrate. What I find very interesting is having the average efficiency since the last reset, so I can see how efficient my miner/setup is. Knowing the live efficiency isn't useful because, in my case, the hashrate varies a lot, from 4.5 to 6.0. But if we have the average efficiency since the reset at 18 J/TH, that's good; if it's 23 J/TH, that's not good. Thank you very much for the work and for responding to suggestions. You would open the ticket :)

shufps, 2025-09-21:
>  in my case, the hashrate varies a lot, from 4.5 to 6.0

But it wouldn't vary that much anymore with the new measurement 🤔 

This are for example values from the measurement:

4887.474, 4868.12, 4923.604, 4836.285, 4830.698, 4894.786, 4913.281, 4867.264, 4879.302, 4856.494, 4851.344, 4894.365, 4812.633, 4890.48, 4838.44, 4928.348, 4891.112, 4939.511, 4806.611, 4933.509, 4885.324, 4856.077, 4831.558,
4906.379, 4913.711, 4858.733, 4907.689, 4862.097, 4954.573, 4872.42, 4908.98, 4897.367, 4840.591, 4822.525, 4950.271, 4827.686, 4932.207, 4889.194, 4808.332, 4825.938, 4921.023, 4833.709, 4935.828, 4943.39, 4896.938, 4885.754, 4882.744, 4
845.322, 4854.354, 4969.627, 4848.332, 4922.744, 4915.0, 4906.83, 4819.944, 4873.281, 4846.612, 4834.57, 4764.029, 4896.936, 4888.334, 4928.335, 4901.237, 4977.369, 4840.161, 4928.766

Maybe I will apply a sliiiiiight average filter on top but hmm

alucine, 2025-09-21:
This is easy logic to calculate cumulative average efficiency, also we have expected efficiency, you can compare this two

#include <iostream>
#include <chrono>
#include <thread>

int main() {
    double power; // Power in watts
    double hashrate; // Hashrate
    double total_efficiency = 0.0; // Cumulative efficiency (hashrate per watt)
    int count = 0; // Number of measurements

    std::cout << "Enter power (watts) and hashrate (or non-positive power to exit):\n";

    while (true) {
        // Input power and hashrate
        std::cout << "Power (W): ";
        std::cin >> power;
        if (power <= 0) break; // Exit if power is non-positive
        std::cout << "Hashrate: ";
        std::cin >> hashrate;

        // Calculate efficiency for this measurement
        double efficiency = hashrate / power;

        // Update cumulative average
        count++;
        total_efficiency = ((total_efficiency * (count - 1)) + efficiency) / count;

        // Display current efficiency and cumulative average
        std::cout << "Current efficiency: " << efficiency << " (Hashrate/Watt)\n";
        std::cout << "Cumulative average efficiency: " << total_efficiency << " (Hashrate/Watt)\n";

        // Wait for 1 second before next measurement
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }

    if (count > 0) {
        std::cout << "Final cumulative average efficiency: " << total_efficiency << " (Hashrate/Watt)\n";
    } else {
        std::cout << "No measurements were taken.\n";
    }

    return 0;
}

alucine, 2025-09-22:
Testing new version 1.33.beta-4, amazing new hashrate, in panel real hashrate in graph is increase, i found a little bug, in lcd the effiency has the old calculate, not the panel.

Now i have 5,237 hash rate and 5,222 hash rate expected !!  amazing

kukulle-hood, 2025-09-22:
> Testing new version 1.33.beta-4, amazing new hashrate, in panel real hashrate in graph is increase, i found a little bug, in lcd the effiency has the old calculate, not the panel.
> 
> Now i have 5,237 hash rate and 5,222 hash rate expected !! amazing
> 
> New effiency is 17.42 j/th with 91.9 watt
> 
> More things, with this new real hashrate, is better to test script overclocking or underclocking, the result will be better ajust !!

@alucine May I ask you on what frequency you are?